groupgets/LeptonModule #11: Connecting with STM32F3Discovery

mrmamon, 2015-07-06:
I'm trying to stream image from lepton to PC via STM32F3Discovery board.
but I'm not sure why I got only "resync" from serial.

I've connected cable like your image in the website.

CS->PA6
MOSI->PA4
MISO->PC5
CLK->PA7
GND->GND
VIN->5V

[image]

Evilboyd, 2015-07-06:
Same thing Cornell got

EnkhbatB, 2016-08-05:
Hello, Why did not you use an I2C interface?
Can I use the camera without I2C interface?