groupgets/LeptonModule #13: ThermalView.exe may contain malware

MartyMacGyver, 2015-08-10:
Over a dozen virus scanners are detecting a trojan in ThermalView.exe:

https://www.virustotal.com/en/file/055ac103ab139b6cb81c632fd67e38f1a758704c82683cbebb266bc3fb9a2b8e/analysis/1439180429/

PureEngineering, 2015-08-10:
Thanks for the info, that is weird.. If you recompile from the source do you get the same trojans?

MartyMacGyver, 2015-08-10:
I am not using this module and cannot compile it at this time (it's not clear what environment this requires that I don't already have) - I was alerted to the potential virus when I  cloned this repo, and investigated as above.

I would suggest removing the executable and leave it to the user to build their binary, at least until a clean binary is available or (if this is a false positive) the reason for the problem is more clear.

MartyMacGyver, 2015-08-10:
So, I thought it was fine but it wasn't... still showing the same heuristic detection though not as many. Not sure what it is about the code that is triggering this but it'll trip any antivirus that sees it as a threat.

MartyMacGyver, 2015-08-10:
Well, this ranks up there as one of the weirder things I've ever seen: the problem is the application icon, period. Using a different app icon resolves the issue. I'll create a pull request tonight with that as well as an updated readme and a couple other minor cleanups.

(Background: I'm going to guess that this old generic icon is featured in many a demo app, and probably many a malicious app derived from a demo app. I scanned the icon alone and got nothing... and I scanned the app with no icon and also got nothing. But the combination of the two triggers the heuristics for multiple scanners.)

closed 2015-08-11 by MartyMacGyver