groupgets/LeptonModule #17: "Perform FFC" button in raspberrypi_video No longer works

belsinb, 2015-10-13:
The previous iteration of the repo on my Pi worked fine, so I know I have the correct settings to communicate with my lepton via I2C.

However FFC doesn't work now ever since the github went from github.com/PureEngineering/LeptonModule to github.com/groupgets/LeptonModule. 

Also, I did run the executable with sudo.

kekiefer, 2015-10-13:
It probably wasn't the move but rather subsequent changes made around then (e.g. pull request #14 or #16). Can you verify if this is the case? Maybe try running the code at version a9905796

MartyMacGyver, 2015-10-13:
I'm not aware of any changes I made in #16 that would have caused FFC not to work (I've been using it without issues). Which subprogram are you using (e.g., raspberrypi_qt, raspberrypi_video, raspberrypi_capture)?

MartyMacGyver, 2015-10-13:
Narrowing down, you must be using raspberrypi_video. I just re-downloaded the repo and rebuilt raspberrypi_video (qmake && make) and ran it - commanded FFC (by clicking "Perform FFC") works - I can hear and see the shutter closing and opening.

Did you download the current release to a clean area and rebuild it? Did you make any customizations previously (different I2C port, etc.)? Are you seeing video?

PureEngineering, 2015-10-14:
did you enable the i2c module in the config.

belsinb, 2015-10-14:
Yes, I am using raspberrypi_video.

Yes, I do have the I2C module enabled. 

No, I didn't make any customizations.

I'll try the build at a990579 to comfirm. 

Thanks you all for the tips.

MartyMacGyver, 2015-10-14:
Let us know what exactly isn't working currently - just the FFC button? Something else as well (e.g., no video)?

belsinb, 2015-11-05:
Turns out I was using such an old Lepton that the FFC function wasn't featured yet. Sorry for the confusion guys. haha

closed 2015-11-05 by belsinb

MartyMacGyver, 2015-11-05:
No problem - thanks for the update!