groupgets/LeptonModule #21: resolution

jeremy-rutman, 2015-10-30:
Hi 
We have a FLIR 160x120 pixel variant; where is the resolution defined? 
I see in LeptonThread.cpp and .h a PACKETS_PER_FRAME 60 and PACKET_SIZE 164 which gets divided by two and is thus suspiciously close to 80 - if I change those to 120 and 328 respectively might that take care of it?
I do see the number 60 and 80 hardcoded quite a bit more (e.g. in STM...ChibiOS/flir_dump.c) - are those camera resolutions too?

MartyMacGyver, 2015-10-30:
If you take a look at this investigation into the interface for the 80x60 FLIR (http://electricstuff.co.uk/lepton.html) you'll see the packet is indeed 4 bytes of header followed by 160 bytes of data (two bytes each), and 60 of those make a frame. I can't say for certain if the 160x120 is exactly the same (or where the docs for that might be) but that's how the 80x60 functions.

Now I want one of those 160x120s..... :-D

MartyMacGyver, 2015-10-30:
BTW, that 160x120 sounds like a Tau 2... more docs here (http://www.flir.com/cores/display/?id=54717) and here (http://www.flir.com/cores/display/?id=61914). Hope that helps!

jeremy-rutman, 2015-10-30:
Thanks.  We tried PACKETS_PER_FRAME 120 and PACKET_SIZE 324 as well as 328, nothing works yet.  Now trying the python interface which at least has rows , cols definitions instead of everything hardcoded.

adgriff2, 2015-10-30:
The Lepton you are referring to has a different vospi protocol. It sends
its data as 4 independent segments with the segment number encoded across
the headers of several different packets. I don't believe we've published
the inner workings publicly yet, so I really can't publish any of my code
yet.

adgriff2, 2015-10-30:
I will also add that the Raspberry Pi has latency issues with it's SPI bus
due to the broadcom driver. As a result, it is too slow to reliably pull
data from the Lepton 3.0. One solution would be to write a custom kernel
module to drive the lepton, but a simple FTDI chip can be a much cleaner
setup depending on the application.

kekiefer, 2015-10-30:
You might give this a try: https://github.com/groupgets/pylepton/tree/lepton3

jeremy-rutman, 2015-10-31:
thanks that branch works like a champ.

closed 2015-10-31 by jeremy-rutman