groupgets/LeptonModule #22: nucleo stm32f401re code - higher frame rate?

mechashef, 2016-01-23:
hello,

I've been trying to get higher frame rate with the nucleo stm32f401re code without any succes.
anyone here optimized it and got higher frame rate than one picture every 2 seconds?
I think its a problem with the reading function reading too slow.

thanks

kekiefer, 2016-01-23:
The firmware for the GroupGets [PureThermal 1 Lepton board](https://groupgets.com/manufacturers/groupgets-labs/products/pure-thermal-1-flir-lepton-dev-kit) will probably be of great interest to you... it's based on STM32F411! :)

You can get full 26 fps if you sync just the first line with CPU or interrupt driven transfers. Once you're in sync, you can dump the rest of the frame out to a buffer with a DMA transfer, and do your work as you kick off the next transfer. Take a look here: 

https://github.com/groupgets/purethermal1-firmware/blob/master/Src/lepton.c#L44

closed 2016-01-23 by kekiefer

mechashef, 2016-01-23:
thank you very much! I hope i can work it out, looks prety complicated