groupgets/LeptonModule #23: Parameterize which i2c bus is used on the Pi

pcernek, 2016-03-04:
Currently, the i2c bus to use on the Pi is hard-coded to `/dev/i2c-1` in the file `software/raspberrypi_libs/leptonSDKEmb32PUB/raspi_I2C.c`. For those who want to connect two Leptons to a single Raspberry Pi, it would be better if this were parameterized in this file and specified at a higher level.

kekiefer, 2016-03-04:
For reference, the OP's [Google Groups post on this topic](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!topic/flir-lepton/y6ljXkb0cnA).

Please let us know if you come up with a solution, we'll be happy to merge your pull request.

kekiefer, 2016-03-14:
PR #24 merged

closed 2016-03-14 by kekiefer

wenyueqiang, 2016-05-09:
MY LEPTON IIC have no response!why