groupgets/LeptonModule #26: Cannot qmake raspberrypi_video

AKAMEDIASYSTEM, 2016-04-15:
Hello, I've located what I believe are the most accurate instructions to get the Lepton module running.

I have gone into /raspberrypi_libs/leptonSDKEmb32PUB and run `sudo make`
Then I have gone into /raspberrypi_video and run `sudo qmake && make`

However, I'm unable to get past this step. I get the error "can't create gen_objs/Lepton_I2C.o: Permission denied"

I have confirmed that my Pi is set up to default enable SPI and I2C upon boot.

Thanks for any help you can provide!

AKA

MartyMacGyver, 2016-04-15:
I'm not sure why you're doing it that way (looks like the old way). Try this:

    git clone https://github.com/groupgets/LeptonModule LeptonModule-newclone
    cd LeptonModule-newclone
    cd software/raspberrypi_video
    qmake && make

(I specified `LeptonModule-newclone` just to ensure you are in a clean repo clone for this attempt.)

AKAMEDIASYSTEM, 2016-04-15:
That worked on the first try! Not sure how I managed to insert that extraneous step, but thank you for your quick help!
AKA

closed 2016-04-15 by AKAMEDIASYSTEM

MartyMacGyver, 2016-04-15:
I tend to work off my notes so it might be that you had notes based on the older process from before I simplified it.