groupgets/LeptonModule #27: V4l2 Lepton

raspiuser200, 2016-04-27:
Hello,

I'm using your Lepton module FLIR, and I'm trying to use the v4l2 program for converting module to a dev / videoX. But the result is not satisfactory

I'm using Raspberry (raspbian). I got the headers for the kernel version 4.4.8-v7 + I have, but when installing it gives error.
How did you hit one?

Nevertheless, I tried to compile and run your code, but I get an error v4l2 Sink.

Could you help me? Thank you and congratulations for your work.

marty-sullivan, 2016-04-27:
Hi, were you able to install the v4l2loopback successfully and add it to the kernel with modprobe?

raspiuser200, 2016-05-25:
Hi @marty-sullivan 

Yes, I tried to install v4l2loopback, with several different kernels 4.4. and  4.1.XX-v7 , , but I can't successfully run modprobe v4l2loopback.

If you did a raspberry, you could comment me like?

Because I am referring to the github of v4l2loopback, and I couldn't get it.

Sorry for the inconvenience. And thanks in advance.
 .

marty-sullivan, 2016-05-25:
Here are the steps I took to compile the v4l2loopback kernel module on a raspberry pi:

1. Install kernel headers with [rpi-source](https://github.com/notro/rpi-source) utility.
2. Compile [v4l2loopback source](https://github.com/umlaeute/v4l2loopback)
3. If step 2 is successful, you should have v4l2loopback.ko in the source directory
4. Copy v4l2loopback.ko to /lib/modules (I believe just /lib/modules directory works, but if that doesn't work, put it in /lib/modules/$(uname -r) ).
5. You might find insmod works too, I just copied the file like step 4 says
6. At this point, you should be able to `modprobe v4l2loopback`

Beware that updating kernel means you will have to recompile v4l2loopback by following these steps again.

raspiuser200, 2016-05-25:
Hi @marty-sullivan 

THANKS!

Apparently it was a problem with my kernel and headers used to. I had downloaded from another website but by using the tool you have told me has been fixed (RPI-source).
Thanks for the reply speed. I was very disappointed because I couldn't find the error, and finally works.

closed 2016-05-26 by marty-sullivan

Cullpepper, 2016-12-03:
Hi Marty,

I think I installed everything correctly, but trying to launch v4l2lepton gives me this code:
"Waiting for sink"
"done reading, rests"

And that's it. Any suggestions?

Cherry97Pepis, 2021-07-16:
Hi Cullpepper,
I have the same problem. I'm trying to stream thermal video from Lepton 2.5 through Raspberry Pi 4B to my iOS app. It's kind of prototype of drone with thermal vision. 
When I start the ./v4l2lepton I get "Waiting for sink", sometimes I get sth weird on the screen in VLC, but usually I get nothing :( 
Have you any idea to fix it? 
I try most of ways to fix it, but people say "wow, it working" and I can only say "nope, still nothing change" :P

Cullpepper, 2021-07-16:
Oh heck, that was 5 years ago, I barely remember now. 

Your issue could be totally different but If I recalL, my problem turned out to be the physical housing of the flir in the board- the sensor is only loosely fitted to the contact points and I had to pop it out and re-seat it.

Cherry97Pepis, 2021-07-16:
Oh Thx for reply 
I’m really broken abt my project 
That is my master Thesis and i have two weeks to finish it 

I will try to check the camera as u say. Maybe that will help 
Thanks a lot :)

EdwinNavarro, 2023-04-07:
Hi Cherry,
Were you able to fix your issue? 
I wonder if this still works

Cherry97Pepis, 2023-04-07:
Hi Edwin, 
Yes, I fixed my issue, I had to use v4l2loopback kernel module and Motion program. V4l2loopback helped me to create virtual video device and stream image from camera. My RaspberryPi didn't recognise FLIR camera before so I had no other choice and fast way to stream video. When I created virtual device in v4l2loopback i was able to see the image from my camera in VLC player.
Motion program was used to send video to my iOS App :)
If you have any questions just ask :) It was some time ago, but maybe my master thesis will help :)

EdwinNavarro, 2023-04-11:
I want to do something similar to what you did. However I was planning on writing a gui, and thermal feed would be done by using opencv, then stream it using vnc viewer. I haven’t been able to open the thermal stream using opencv. I need to control some gpio pins as digital outputs. An iOS app would be better though