groupgets/LeptonModule #34: /dev/video1 of v4l2lepton

Jakebo, 2016-09-01:
Dear All,

I downloaded the LeptonModule project and viewed the source code of v4l2lepton.
Now I am confused by the device node "/dev/video1", How is video1 come from? Is generated by Lepton driver? And where is the driver?

Can you provide some helps?
Thanks in advance,
Jakebo

marty-sullivan, 2016-09-01:
The /dev/video1 device is the default name of the loopback device that is created by v4l2loopback, and the default device the v4l2lepton looks for. 

v4l2lepton requires that the v4l2loopback kernel module be installed. You can specify when inserting the v4l2loopback module which /dev/videoX device is created, either with modprobe or in /etc/modules (see v4l2loopback docs for how to specify this)

If you decide to use a different device number (for example, maybe you have multiple loopback devices or multiple cameras attached to your Pi) then you can specify the device name for your Lepton when running v4l2lepton. For example, if you create a v4l2loopback device named /dev/video2, you can run `./v4l2lepton /dev/video2` and it will use that device instead of /dev/video1.

Jakebo, 2016-09-06:
Thanks marty-sullivan, this is great helpful for me.