groupgets/LeptonModule #37: Waiting for sink - resets

Cullpepper, 2016-12-03:
Hi, 

I've been trying to get the v4l2lepton module running and I get the message below - any suggestions? I'm not sure what it's telling me.

```
pi@raspberrypi:~/LeptonModule/software/v4l2lepton $ ./v4l2lepton /dev/video0
Waiting for sink
done reading, resets: 

```

marty-sullivan, 2016-12-03:
Hi, I think you can probably ignore that message. Or is it not streaming to the loopback device?

Cullpepper, 2016-12-06:
Hi Marty, 

Sorry, no. I was never able to get it working.

marty-sullivan, 2016-12-06:
OK, I will need to know more about your setup in order to help. 

A good place to start is to see if the plain raspberrypi_video works on your Pi. This will confirm the breakout board and Lepton are correctly connected and able to stream frames.

Assuming that works, the next step is to confirm that v4l2loopback module is installed correctly and streaming frames properly. Since you aren't getting the message "Failed to open v4l2sink device" I feel like this is probably working.

After you've started v4l2lepton, use v4l2-ctl to confirm that your `/dev/videoX` loopback device has the proper configuration. It should be streaming 80x60 BGR frames (maybe RGB, can't remember).

I suppose another question would be, how are you trying to access the loopback device to see if the frames are streaming? An easy test would just be to use VLC media player to open the stream.

spacemuff, 2017-01-18:
Hi,

having something similar.
The raspberry_video works, I have the Qt window open and can perform FFC etc etc.
If I run the modprobe v4l2loopback it creates the dev/videoX
but if I execute ./v4l2lepton /dev/video0 I get the same error message.
If I try -ctl I get nothing

Thanks in advance

s

spacemuff, 2017-01-19:
OK I have collected some data, hope it helps (trying to help me...)

- no "done reading, resets:" msg only if I unplug the CS cable
- CS cable only seems to work if i plug it to the SPIO CS1 instead of CS0 port of the RPi3 (CS0 is stated in the original description)
- I am using scorpio breakout board (http://www.0gravity.ch/online//product_info.php?manufacturers_id=14%C3%82%C2%A0osCsid%3Dd5a87189f69a74bacb8755fc719f3452%C3%82%C2%A0osCsid%3D50f4bf84ad4f518a50a0833ad3dc1e0a&products_id=1610&language=de)
- v4l2 --all
Driver Info (not using libv4l2):
	Driver name   : v4l2 loopback
	Card type     : Dummy video device (0x0000)
	Bus info      : platform:v4l2loopback-000
	Driver version: 4.4.38
	Capabilities  : 0x85208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x85208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
Priority: 0
Video input : 0 (loopback: ok)
Video output: 0 (loopback in)
Format Video Output:
	Width/Height  : 80/60
	Pixel Format  : 'RGB3'
	Field         : None
	Bytes per Line: 240
	Size Image    : 14400
	Colorspace    : SRGB
	Flags         : 
Streaming Parameters Video Capture:
	Frames per second: 30.000 (30/1)
	Read buffers     : 8
Streaming Parameters Video Output:
	Frames per second: 30.000 (30/1)
	Write buffers    : 8

User Controls

                    keep_format (bool)   : default=0 value=0
              sustain_framerate (bool)   : default=0 value=0
                        timeout (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io (bool)   : default=0 value=0
                    keep_format (bool)   : default=0 value=0
              sustain_framerate (bool)   : default=0 value=0
                        timeout (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io (bool)   : default=0 value=0

Thanks in advance

marty-sullivan, 2017-01-19:
Thanks for the debug info. I will hopefully have time to look at this over the weekend. I want to keep this code usable because I really think it is the best way to interface with the Lepton on RPi with the basic breakouts.

marty-sullivan, 2017-01-25:
Starting from the latest Raspbian image (with GUI) on Raspberry Pi 3 model B v1.2 with the PureEngineering breakout board / Lepton 2 wired. I had keyboard and mouse connected, and I did the following steps:

1. I ran raspi-config to enable all interfacing (i2c, spi)
2. `sudo apt-get update && sudo apt-get dist-upgrade` & rebooted
3. `sudo rpi-update` & rebooted
4. I installed and ran `rpi-source` 
5. I built v4l2loopback from source, copied v4l2loopback.ko to /lib/modules & /lib/modules/$(uname -r)
6. `sudo depmod -a`
7. `sudo modprobe v4l2loopback`
8. At this point /dev/video0 is ready
9. I built LeptonModule/raspberrypi_capture and ran to confirm that the Lepton was wired properly and able to capture frames.
10. I built LeptonModule/v4l2lepton
11. `./v4l2lepton /dev/video0`
12. I receive the message `Waiting for Sink` and then `done reading, resets:` every time the shutter closes.
13. I installed and ran VLC Media Player. I opened the video capture device, chose /dev/video0 and opened it
14. VLC successfully showed me the frames streaming from the Lepton.

Please follow these steps and see if you are still having problems.

spacemuff, 2017-01-28:
Hi,

I could make it work by following your steps (plus some fiddling around)

Plus I have noticed, that sometimes the camera needs some help by *disconnecting then reconnecting the SPIO CS0 cable while ./v4l2lepton /dev/video0 is running*

most of the cases it solves the stuck in waiting for sink problem

marty-sullivan, 2017-01-30:
I have a feeling that the problem is similar to the "red square" problem with the raspberrypi_video program. The code is mostly the same for v4l2lepton, and I did notice that I was having problems running raspberrypi_video when I went back and checked but not with the other sample programs. Not really sure what the fix is other than playing with the wiring like you have done.

Cullpepper, 2017-01-30:
Don't forget to check the physical mounting of the lepton. After endless debugging, I noticed the lepton was loosely fitted in the breakout board and would slip, causing the red square. 

That might not be your problem, but it's a lot easier to check and eliminate as an issue!

clint3501, 2017-04-06:
Marty's guide above in this post works.    I just changed the video0 to video1 since the pi camera likes video0 with the modprobe v4l2loopback video_nr=1 

vlc works with this example and is on the pi itself already

kevin192291, 2018-01-20:
Hello, I am trying to get this to work, but I keep getting "Failed to open v4l2sink device.", however if I run pyLepton everything works great. I am have tried both CS0 and CS1 on the PI Zero, I would like to eventually get this streaming data to a webpage. Any help would be greatly appreciated.

Cullpepper, 2018-01-20:
If you figure this out, can you please post your configuration? I tried the same thing for a while and never got it to work. I think I failed in getting the vl42 dependencies installed correctly.

marty-sullivan, 2018-01-20:
The problem is certainly due to your v4l2loopback configuration. My suggestion is to spend time trying to get v4l2loopback to work with a normal webcam first to identify the problem. If you follow the steps exactly above, it should work. Although, those steps were done on raspbian-jessie and I haven't tested on stretch, so you may want to use jessie.

davidbendell, 2018-01-28:
I have spent hundreds of hours trying to get this working. Everything is hopeless and frustrating. I have followed your steps above but whichever route I take, I keep hitting the same problem:
`pi@raspberrypi:~/Downloads/leptonmm/software/v4l2lepton $ ./v4l2lepton /dev/video1
Waiting for sink`

Any suggestions from anyone on how to get this working?

marty-sullivan, 2018-01-29:
Glad you got it working!

dmoreyes, 2018-02-07:
Hi everyone, does this driver work with Lepton 3 model also ? I believe it's not due to resolution differences but I would love to be wrong.

marty-sullivan, 2018-02-07:
As it is right now, it only supports Lepton 2. I don't have a Lepton 3 to test, but it could be as simple as changing the any of the relevant constants highlighted in the link below to match the new resolution and recompiling.

https://github.com/groupgets/LeptonModule/blob/1bc9aa08622e9b686c2ec90b95fa3dedd09d9166/software/v4l2lepton/v4l2lepton.cpp#L25-L37

I think I remembered reading at one point that the SPI interface on Raspi might not support the bandwidth needed for the Lepton 3, but that might have either changed or maybe the new Pis have improved since back then. It's worth a shot either way!

dmoreyes, 2018-02-07:
Thank you for your answer, this is something I could definitely test! I can change resolution parameters to 160x120 as a start but the most tricky part is probably around the need to read and drop fake images before getting the right one. Not sure if Lepton 2 works the same way.

marty-sullivan, 2018-02-07:
FYI, it looks like the solution to the SPI issue is simply increasing the buffer size in the Linux kernel. You can set this in /boot/cmdline.txt:

spidev.bufsiz=131072

adgriff2, 2018-02-07:
I haven't worked with lepton 3 on a pi since 2014 and at that time the
kernel was the bottleneck for the SPI interface. There was a significant
overhead for each read call. If you tried to read in a buffer that was
large enough to hold a full row of pixels, you'd incur the overhead on
every row and not get the full frame before the next frame refreshed. This
wasn't a problem on lepton 2 variants because it was 60x80 pixels, but
became an issue with 120x160 pixels. The easy solution I found was to have
a large buffer and read a full segment at a time instead of row at a time.

The 60x80 image is transmitted as a single segment, but the 120x160 image
is transmitted as four 30x160 segments that need to be reassembled. This
segmentation difference is probably the most likely complication you would
run into by trying to just change any hard-coded 60 and 80 integers to 120
and 160 respectively.

As i understand it, the pi SPI overhead was a software issue that was
planned to be fixed in later kernels, so i would hope that has been
released by now.

adgriff2, 2018-02-07:
Here is some very old code (warts and all) that pulls the lepton3 image and serves it over tcp/ip to a connecting client that I wrote in C#.

Only the lepton2 version of the client/server code is published on github [here.](https://github.com/adgriff2/LeptonPi) I believe i couldn't publish lepton3 code at the time because it wasn't released yet.

```
#include <stdio.h>
#include <sys/types.h>
#include <sys/socket.h>
#include <netinet/in.h>
#include <netdb.h>
#include <string.h>
#include <stdlib.h>
#include <unistd.h>
#include <errno.h>
#include <linux/spi/spidev.h>
#include <sys/ioctl.h>
#include <fcntl.h>
#include <pthread.h>
#include <iostream>
#include <time.h>
#include <limits>
#include <arpa/inet.h>
#include <linux/i2c-dev.h>
#include "SDK/LEPTON_SDK.h"
#include "SDK/LEPTON_SYS.h"
#include "SDK/LEPTON_Types.h"
#include "SDK/LEPTON_I2C_Reg.h"

// Constants that should never change
#define HEIGHT 120
#define WIDTH 160
#define BYTES_PER_HEADER 4
#define BYTES_PER_PIXEL 2
#define SEGMENTS 4
#define SEGMENT_FACTOR 2 // sqrt(SEGMENTS)
#define SEGMENT_LINES (HEIGHT/SEGMENTS)

// I2C constants
#define I2C_ADDRESS  (0x2A)
#define AGC (0x01)
#define SYS (0x02)
#define VID (0x03)
#define OEM (0x08)

#define GET (0x00)
#define SET (0x01)
#define RUN (0x02)

#define STATUS_REG (0x02)
#define COMMANDID_REG (0x04)
#define DATALEN_REG (0x06)
#define DATA0 (0x08)

// Maximum values for array allocations
#define PACKET_SIZE (BYTES_PER_HEADER + WIDTH * BYTES_PER_PIXEL/SEGMENT_FACTOR)
#define FRAME_SIZE (HEIGHT * WIDTH)
#define SEGMENT_PACKETS (SEGMENT_LINES * SEGMENT_FACTOR)
#define SEGMENT_SIZE (PACKET_SIZE * SEGMENT_PACKETS)

using namespace std;
pthread_mutex_t lock;

void message(const char *msg){
	pthread_mutex_lock(&lock);
	time_t t = time(NULL);
	struct tm tm = *localtime(&t);
	char time_string[200];
	sprintf(time_string,"[%4d-%.2d-%.2d %.2d:%.2d:%.2d] %s\n", tm.tm_year + 1900, tm.tm_mon + 1, tm.tm_mday, tm.tm_hour, tm.tm_min, tm.tm_sec, msg);
	cout << time_string;
	pthread_mutex_unlock(&lock);
[... 447 lines cut ...]
			break;
		}
	}
}

int main(int argc, char *argv[]){
	pthread_t LeptonRead_thread, ImageSocket_thread, CMDSocket_thread;
	pthread_mutex_init(&lock, NULL);
	message("Starting LeptonPi2 Server!");
	pthread_create(&LeptonRead_thread, NULL, handleLeptonRead, (void *)1);
	int sockfd_listen = ListenSocket(5000);
	int newsockfd_image, newsockfd_cmd;
	char buffer[500];
	sprintf(buffer,"HEIGHT: %i\nWIDTH: %i\nBYTES_PER_HEADER: %i\nBYTES_PER_PIXEL: %i\nSEGMENTS: %i\nSEGMENT_FACTOR: %i\nSEGMENT_LINES: %i\nPACKET_SIZE: %i\nFRAME_SIZE: %i\nSEGMENT_PACKETS: %i\nEGMENT_SIZE: %i\n",HEIGHT,WIDTH,BYTES_PER_HEADER,BYTES_PER_PIXEL,SEGMENTS,SEGMENT_FACTOR,SEGMENT_LINES,PACKET_SIZE,FRAME_SIZE,SEGMENT_PACKETS,SEGMENT_SIZE);
	message(buffer);
	for (;;) {
		GetSocket(sockfd_listen, &newsockfd_image,&newsockfd_cmd);	
		pthread_create(&ImageSocket_thread, NULL, handleImageSocket, &newsockfd_image);
		pthread_create(&CMDSocket_thread, NULL, handleCMDSocket, &newsockfd_cmd);
	}
	pthread_join(LeptonRead_thread, NULL);
	message("Lepton Read Exiting");
	exit(-1);
}
```

dmoreyes, 2018-02-07:
Thank you all for your answers ! It really helps ! Just to give a piece of context, at the moment I'm using pylepton Python package. To make it working fine on a quite loaded RPi, I had to dedicate an entire core for my Lepton 3 image acquisition program to avoid being interrupted by other processes. Otherwise I'm experiencing plenty of de-sync, which blocks the Lepton after 10 minutes. That's the reason why I'd like to switch to C/C++ and maybe to preempt-rt kernel to avoid core dedication. Just curious, have you tried this driver under heavily loaded RPi ? If says so did you have to dedicate a core to your application (might be difficult for a driver by the way...) ?

Cullpepper, 2018-02-07:
Hi- I'm out of my depth for a lot of this, but I wanted to ask what you're using to interface with your pi- (virtual desktop, hmdi plug, etc.) on such a small device, the overhead from different configs makes a real difference!

marty-sullivan, 2018-02-07:
Sure thing, and yes thanks for your code @adgriff2 

I haven't done any real performance tests, but I know that I've taken a look at htop while testing v4l2lepton. I recall there being constant CPU usage while it's streaming frames, but it did not take up a full core. My mind is pointing to about 20% of one core, but I don't trust my memory well enough to guarantee that :) 

Of course, you will just have to test yourself, and this tax on the CPU will likely increase with the Lepton 3 considering it has 4x the pixels.

adgriff2, 2018-02-08:
I've not experimented with reading the lepton with pi under heavy processor
load, though I can imagine there would be an issue. It may be possible to
reduce the CPU load of reading the lepton by only reading unique frames.
The lepton output images at a rate of 27Hz to work with standard protocols,
but only refreshes the images internally at 9Hz due to ITAR restrictions.
Basically, unique frame is sent out 3 times to get from 9Hz to 27Hz. You
really only need to read each unique frame once and sleep during the other
2 frames, but i've never bothered to do this.

dmoreyes, 2018-02-09:
> 704-221-1988 <(704)%20221-1988>

Unfortunately, as per the documentation, you need to read every frame,
including invalid ones, otherwise sync is lost :

"For both Lepton and Lepton 3, each unique video frame is followed by two
non-unique frames which must be read out to maintain synchronization. For
Lepton each unique video frame is duplicated twice. For Lepton 3 each
unique frame is followed by two partial, invalid frames."
https://groupgets-files.s3.amazonaws.com/lepton/Lepton-3-Engineering-Datasheet.pdf

adgriff2, 2018-02-09:
> Engineering-Datasheet.pdf
> 704-221-1988 <(704)%20221-1988> <(704)%20221-1988>

Yes, the non-unique frames need to be read. But if you have other overhead
such as displaying video, re-transmitting it over TCP/IP, or encoding it as
a video you can detect non-unique images and potentially avoid resource
intensive post-processing. Again, I haven't looked into this. If merely the
act of reading the vospi constitutes 90% of your load, these tricks won't
help much at all.

dforsberg55, 2019-08-05:
I am currently using a FLIR Lepton 2 with an Odroid XU4, I have set everything up as @marty-sullivan suggested. However, I am getting "waiting for sink", and it just hangs here endlessly.  It never moves onto "done reading, resets:" and I can not see any feed on VLC. 

My camera does work since I can grab single images using PyLepton. I have also managed to stitch images together to create a "slide show". The problem is that I need a video output that can be used in OpenCV, e.g. "cv2.VideoCapture(0)".