groupgets/LeptonModule #43: v4l2lepton and color palette modification

clint3501, 2017-04-06:
I need a little direction on how to adjust or rather modify the color palette for temperature that the lepton module displays.  

I got everything streaming using v4l2 with multiple targets with a raspberry Pi 3B using vlc and also embedding it in a webpage.

However I am having an issue with washing out (wrong term).  

The palettes.cpp has several references and I believe the v4l2lepton code is using ironblack as the reference.

More to the point,    I would like to keep a grey scale (if possible) for temps below  85-90 degrees F and have a single color displayed between 90-120 and then one other color between 120-200 and then another color for those readings above 200 degrees F.     (kelvin in in C and not F I know,  I am using this as an example)
I am not hell bent on the above ranges however this might help the very few who might know the answer in providing me with some guidance on this topic.

I also suspect I can change the reference from ironblack to one of the other two in the cpp file by just changing the lepton.cpp file   from const int *colormap = colormap_ironblack to rainbow or greyscale   which is in the Palettes.cpp file.  I think its one mod to the file however one is never sure and knowing this would help.

Assuming I am correct with the above I have a gap in understanding how to either modify the Palettes.cpp file (I know a text editor) for RELATIVE temp ranges I am interested in.
Presently I have pulled the files from the Palettes.cpp file and noticed there are 3 entries for the 256 temp/colors.  What each one of the fields are in the ironblack or rainbow or greyscale would go a long way for helping me.

A important point is that I am not after an absolute temperature related to the palette color map, just relative and I believe it is set up right now.  

Any actual suggestions, examples, on how to adjust the palette color map is what I am probably after.
If there is a better or different way I would also enjoy this assist as well

Thanks in advance

marty-sullivan, 2017-04-06:
I believe the logic is converting each pixel from the Lepton from 14 bits to 8 bits, taking into account the min and max values (range of the scene) and mapping them to a value between 0-255. Then it simply pulls the corresponding value from the palette for each 8-bit pixel.

So, in other words, 0 will be the minimum temperature in the scene and 255 will be the maximum, so it is *always* relative to the range of temperatures in the scene.

With that being the case, you would have to change the logic in the code to do something different with the 14-bit values in order to map them to the color map you want to use.

Sounds like a cool enhancement if you can get it to work!

And yes, if you change the name of the palette being used in the v4l2lepton source and recompile, it will then use the other color map :)

marty-sullivan, 2017-04-06:
I've highlighted the relevant code that finds the min / max values and then maps them to 8-bits. The 8-bit value is then the index of the color within the colormap. I hope this helps:

https://github.com/groupgets/LeptonModule/blob/8846b7406e2f617fec285f4e9206e247f33cf8c8/software/v4l2lepton/v4l2lepton.cpp#L67-L109

clint3501, 2017-04-07:
thanks, will let you know.