groupgets/LeptonModule #46: Changing Colour Palette for raspberrypi_capture

Jessman101, 2017-08-01:
Is there a way to change the colour palette for the output images from the raspberrypi_capture function?

By default the images are greyscale but I would prefer the images to be rainbow palette or ironblack.

Cheers Jess.

tmsd2001, 2018-04-15:
I think that it is possible.
My question is, has someone finished that?
Or must I reinvent the wheel.
An Example, i have a pgm pic with min 7804 and max 8305, so the pgm has 501 gray values.
Form a Rainbow you have this RGB
First step 000 000 255
then         000 255 255 = 255 Steps
next         000 255 000 = 255 Steps
next        255 255 000 = 255 Steps
end         255 000 000 = 255 Steps
Sum = 1020 Steps
Convert the 501 Values to 1020, with faktor 2,035928144.
Have your pixel a Value 143 * 2,035928144 = 291,1 = round 291.
So RGB ist is 000 255 219
Now you can write a ppm File.

tmsd2001, 2018-04-15:
A other way is use an extern command, i test the convert command from imagemagick.
A tutorial can be found here [http://www.imagemagick.org/discourse-server/viewtopic.php?t=27377](url)