groupgets/LeptonModule #51: Would anyone want a verified working Lepton 3 160x120 MJPEG capture utility that streams to /dev/stdout on Raspberry Pi?

The-Oracle, 2018-03-15:
Using the Raspberry Pi libs from this repo, I wrote a C++ Console application that pipes a MJPEG stream to /dev/stdout. I personally used my utility to pipe feed FFMPEG to live stream. It uses the same color pallet as the Qt GUI Raspberry Pi apps. If anyone is interested, I'll create a public repo of it on Github, or I can provide the source to Group Gets.

marty-sullivan, 2018-03-15:
I'd be interested to update v4l2lepton to support the Lepton 3 on RPi. I don't personally have time to work on it until summer though. Either way your code sounds useful. I'd accept a pull request that adds a switch to support both 160x120 and 80x60 to [v4l2lepton.cpp](https://github.com/groupgets/LeptonModule/blob/master/software/v4l2lepton/v4l2lepton.cpp) and/or a pull request that just adds the code you have made to a separate directory under software.

The-Oracle, 2019-10-18:
I added a switch for supporting 160x120 and 80x60. I am not sure how to add a pull request.

PavelasV, 2019-12-10:
i am trying to use 160x120 lepton on rpi4, but unsuccessfully. Can someone explain how to get a 160x120 stream from breakout board 1.4 with RPI?

kekiefer, 2019-12-10:
Check out #66 -- note the end of the thread, you have to disable frequency scaling on rpi 4

The-Oracle, 2020-11-25:
I'm unsure how to do a pull request. I would like to share my code with the community. I have made some improvements since my last post, I created a separate thread for jpeg encoding, so capture happens on one thread and jpeg compression on a separate thread. I added functions for radiometry spot meter temperature readings for the Lepton 3.5, but the function is unused in the capture utility. I added a few new color palettes: rainbow, greyscale, iron black and glowbow, however fusion is the default color map. There is a command line parameter to select between 160x120 and 80x60 resolutions. There is a command line parameter for selecting the v4l2 loopback device and the chip select pin.

crisposada, 2020-11-26:
@The-Oracle Sounds great! I would appreciate it very much. Do you have the code in another repository?