groupgets/LeptonModule #54: Lepton 3 support?

crazymao, 2018-04-26:
Is this library meant to work with the Lepton 3 module? I am unable to get a video feed from it, no matter what I do I get the red square error. Thanks

israelshirk, 2018-04-29:
There are several pull requests which function properly with lepton 3; I think the original authors are concerned with backwards-compatibility to earlier rpi, regarding an SPI device change.

kekiefer, 2018-04-30:
You're right that there are a few outstanding PRs that are still out for review. If anyone has a working setup, we would be curious to hear about your experiences with the contributed code.

crazymao, 2018-04-30:
I was able to use the code from PR #31, it worked beautifully out of the box, it was the only thing I could find that actually worked.

kekiefer, 2018-04-30:
Did you happen to try #33, with the 20MHz clock?

crazymao, 2018-04-30:
Not yet but I'm planning to, I'll post the results when I do.

TefaTBob, 2018-05-03:
I couldn't get #33 by @amyers-novacoast to work, something must be off in their branch or how I'm compiling it, but this did work for me and my Lepton3: https://github.com/TradeCrafter/lepton3 
It's a little glitchy/jumps into fullscreen if you don't change that, but works well enough to test at least. They created a ton of colormaps and added a screenshot mode too which is neat.