groupgets/LeptonModule #60: v4l2lepton support Lepton3?

qingshutie, 2018-08-29:


marty-sullivan, 2018-09-26:
I would love to make v4l2lepton work with the Lepton3 but I simply don't have the cycles to do so right now, nor do I have a Lepton3 to test with. If I do find the time, I will let you know in this issue. 

If you want to continue trying to update the code yourself, you could try posting your progress here and I may have some pointers to get it working. If you do get it working, I'd also be happy to accept a pull request.

israelshirk, 2018-09-26:
Hey @marty-sullivan, I’d be glad to make the requisite changes and have the hardware; my concern is that it wouldn’t get merged in (like the other PRs for Lepton 3 support with no comments).  I’m instead committing my time to Google’s periph.io project.

If there’s a strong reason to believe I wouldn’t be wasting my time I’d be glad to contribute - just let me know what the gating concerns are.

marty-sullivan, 2018-09-26:
@israelshirk I'm not sure what the pull requests in the pipe are for, but they aren't for v4l2lepton. My only commitment as a collaborator is v4l2lepton, so I won't merge to any other (or new) software contributions. 

If you are willing to update v4l2lepton to support the Lepton3 (and continue to support the Lepton2) then I would be happy to merge, as has been done for previous updates to v4l2lepton.

israelshirk, 2018-09-26:
Ok, I'm going to pass on this.

For those interested, @da-nie has a nice working fork for your Lepton 3 with the PR submitted for this repo, and I'll be maintaining a Go variant as well based on @maruel's previous work.

marty-sullivan, 2018-09-26:
Ok... not sure why you brought it up then. It looks like the PR @da-nie put in is for an adaptation of the raspberrypi_video code that supports Lepton 3. It looks like a good effort and the code can probably be used to update v4l2lepton. I like that it works, but a QT app for viewing the frames from the Lepton is not going to be as useful as enabling V4L2 support for the Lepton 3. 

Apologies if I've misunderstood what you are trying to accomplish.

maruel, 2018-09-26:
Paging @mjs and @clare who worked on the https://github.com/TheCacophonyProject/lepton3. I refactored https://periph.io/x/periph/devices/lepton so that it's now doable to support the lepton3 natively. It just needs to be moved over.

spk921, 2019-05-05:
@qingshutie Have you figured it out?