groupgets/LeptonModule #61: 'VIDEOIO ERROR: V4L2: Could not obtain specifics of capture window' when using OpenCV to open Lepton

dakodako, 2018-09-21:
I tried to run v4l2lepton to make the lepton as one of the devices that can be opened by OpenCV. Everything ran smoothly until I tried to open the device in OpenCV.
I got an error called "VIDEOIO ERROR: V4L2: Could not obtain specifics of capture window"
When I ran the command ls -hl /dev/video*, it gave me 
`crw-rw----+ 1 root video 81, 0 Sep 22 04:45 /dev/video0 `

which I think means that v4l2lepton has been successfully executed.
The command I used to ran v4l2lepton is
`./v4l2lepton -v /dev/video0 -d /dev/spidev0.1`

I am new to OpenCV and Linux. May I ask anyone knows why the error occurred? Thank you very much!

marty-sullivan, 2018-09-21:
Is the problem only with OpenCV using the video stream? Or does trying to open `/dev/video0` as a capture device with VLC media player also fail?

dakodako, 2018-09-25:
Thank you very much for your reply! I successfully got it running now. I realized that it is my fault that I forgot to keep v4l2lepton running when reading the camera from OpenCV.

qingshutie, 2018-09-25:
The v4l2lepton does not work on lepton3,can you help me?

closed 2018-09-26 by marty-sullivan