groupgets/LeptonModule #62: Red Box problem

kapoorshivam100, 2018-10-15:
With the command sudo ./raspberrypi_video only red box is visible and i have tried many fixes but nothing helps..please can someone help me.

Henrikcox, 2019-11-07:
I've been working on the same issue for about a week now with no luck, but here's what I've tried. Let me know if anything here works for you.

- Start easy: check your wiring to make sure everything is connected correctly between the Lepton's breakout module and the pi's pin layout
- Carefully remove and put back the Lepton module from its breakout board. Should hear a click if it's all the way in.
- Play with the CS pin in #24 (CE0) and #26 (CE1). Make sure it matches correspondingly for spidev in the LeptonThread.cpp script in the raspberrypi_qt folder. If your CS pin is in #26, line 14 should read "/dev/spidev0.1" and if you're using #24, it should read "/dev/spidev0.0"
- Experiment with the full SPI pin layout between these two patterns (I did this because I'm considering daisy-chaining with a different module or using SPI0 and SPI1 for each module.
     - [SPI0] MOSI: #19, MISO: #21, SCLK: #23, CE0: #24, CE1: #26
     - [SPI1] MOSI: #38, MISO: #35, SCLK: #40, CE0: #12, CE1: #11
- Set DEBUG_LEPTON to true in the Lepton_thread.h script (also in the raspberrypi_qt folder). If that doesn't help, try setting it back to false but comment out the debug defines surrounding line 48, and changed 'sequence' to 'sequencex' in line 14
- Change the SPI speed to 18000000 from 10000000 in SPI.cpp (under raspberryp_video folder). There will be two places to change that number. 
- Check the voltage through the Lepton breakout board with a voltmeter while plugged in to make sure it isn't fried.
- If you have an oscilloscope or analog reader, check to see what data is being read by the Lepton while it is running raspberrypi_video. This will see if the sensor itself is broken or mis-wired.

Run through each of these one at a time. Hopefully one of them works for you

hanifizzudinrahman, 2019-12-09:
hello henrick, please help me

bossardme, 2021-05-04:
Did anyone figure this out?