groupgets/LeptonModule #64: Very low quality and intermittent images

xzheng77, 2018-11-21:
After downloading and compiling all the code for Lepton 3.5 thermal camera, I ran raspberrypi_video, it was able to output the video, but the resolution of video was very bad, and intermittent. Any idea what cause the problem? And how to fix it. 
PS: I am new to raspberry pi.

pasmitchell21, 2019-03-05:
> After downloading and compiling all the code for Lepton 3.5 thermal camera, I ran raspberrypi_video, it was able to output the video, but the resolution of video was very bad, and intermittent. Any idea what cause the problem? And how to fix it.
> PS: I am new to raspberry pi.

Did you figure this out?

veskoch, 2019-09-10:
Same issue. The image appears "interlaced" (see attached).
[image]

This is on a Raspberry Pi 4, with Flit Lepton 3.5, using the standard breakout board by GetLab.

I see that in `main.cpp` the resolution is set to 320x240, and the Lepton is outputting 160x120 (link below). Can the issue be linked to that?

https://github.com/groupgets/LeptonModule/blob/1bc9aa08622e9b686c2ec90b95fa3dedd09d9166/software/raspberrypi_video/main.cpp#L26

kekiefer, 2019-09-10:
This code is not compatible with the Lepton 3. That is why the image looks like this, it's actually 80x60 frames.

closed 2019-09-10 by kekiefer