groupgets/LeptonModule #67: Get temperature from sensor

frank1789, 2019-09-26:
Hi, I'm currently working with a FLIR lepton 2.5 sensor.
My software is based on [raspberrypi_video](https://github.com/groupgets/LeptonModule/tree/master/software/raspberrypi_video) and works well on the Raspberry Pi 3b, I would like to know if there is any way to generate a relationship between the image and the maximum and minimum temperatures.
I currently get from the functions:
```C
LEP_GetSysFpaTemperatureKelvin(...)
```
* get 30511 (305.11 K)

```C
LEP_GetSysAuxTemperatureKelvin(...)
```
* get 30460 (304.60)

both should be expressed in Kelvin.
I look forward to suggestions and thank you in advance.

You can view my [source](https://github.com/frank1789/Flir-Lepton)

hanifizzudinrahman, 2019-12-06:
Hello frank1789, 
i didnt test your code, but as i see, your code is in C programming language.
do you have the code for python, because i run it in raspberry pi B?

frank1789, 2019-12-06:
Thanks for the reply, unfortunately I have no python code available to carry out the tests.
I also forgot to specify to use branch _"feature/temperature"_ from git repository.

hanifizzudinrahman, 2019-12-11:
i use the code from Frank
[LEPTON 2.zip](https://github.com/groupgets/LeptonModule/files/3948054/LEPTON.2.zip)

But, there is little problem in my project (**Lepton 2.5** with board V1.4 + **Raspberry Pi 3 B**)
my schematic is

- CS = SPI0 CS0 (Pin 24)            # **Use spidev0.0**
- MOSI = SPI0 MOSI (Pin 19)
- MISO = SPI0 MISO (Pin 21)
- CLK = SPI0 SCLK (Pin 23)
- GND = GND (Pin 14)
- VIN = 3.3V PWR (Pin 17)
- SDA = I2C1 SDA (Pin 3)
- SCL = I2C1 SCL (Pin 5)

According with the suggestion above, i make some change in ....
*** LeptonThread.cpp**
- const char *LeptonThread::device = "/dev/spidev0.0";   # (can change in spidev0.1)
- unsigned int LeptonThread::speed = 18000000;            # (from 16000000)
- qDebug() << resets << "reset," << errors << "errors";    # (before this line is commented)
*** LeptonThread.h**
- #define DEBUG_LEPTON true                                       # (before this line is false)

when i run the Lepton-CaptureRaw, the result is good like this, but... 
[image]

but..., sometimes when i try **10 times run this** i can video it the thermal video and when i press the button "Snapsot" it will give me the .jpg picture (1-rgb.jpg, 2-rgb.jpg) and .bin format (1-raw.bin, 2-raw.bin) that this raw format will show a temperature value (i'm not yet with this) but **2 times will fail**. 
The error is **"Error transfering SPI packet"** and the **layer full of red**. Can you help me with this? why this happen?

frank1789, 2019-12-11:
project (Lepton 2.5 with board V1.4 + Raspberry Pi 3 B)
I use the following:

| Raspberry GPIO  | Breakout Board |
|:----------------|-----------------:|
|(Pin 1) +3.3V| VIN|
|(Pin 3) SDA | SDA|
|(Pin5) SCL  |  SCL|
|(Pin6) GND |GND|
|(Pin 19)MOSI |MOSI|
|(Pin 21) MISO| MISO|
|(Pin 23) SCLK  |  CLK|
|(Pin 24) CE0 |CS|

+ This configuration works well and avoids the red rectangle on the screen most of the time;
+I preferred a method that would automatically decide whether to use:
 _"/dev/spidev0.1"_ or _"/dev/spidev0.1"_;
+ I can't explain why after red sessions I can't explain, but it is my opinion that some camera register or on the breakout board is not reset correctly and that it retains memory. Or some counter overflows. To solve this problem, just disconnect the camera from the socket and re-insert it after 10[s].

[image]
[image]