groupgets/LeptonModule #70: Red square on black screen - raspberrypi_video

Henrikcox, 2019-11-07:
I've been trying to set up a Lepton Thermal Camera (breakout v1.4) with both a raspberry pi 3 and a zero W. I started with this wiring and setup guide and have been troubleshooting with all the suggestions mentioned in this thread with no luck.

Running `sudo ./raspberrypi_video `
   - Black pop up window with red square in the upper left corner. 'Perform FCC' button works and causes the Lepton camera's shutter to close and open. 

Running `sudo ./raspberrypi_capture [with output file name]`
   - Message 'spi mode: 0   bits per word: 8   max speed: 16000000 Hz (16000KHz)', terminal hangs on a blank command line. Ctrl-c to quit

Running `sudo ./raspberrypi_overlay [with options such as '-a 155']`
   - terminal hangs on a blank command line (pi camera IS connected for this one)

Running `sudo ./raspberrypi_qt`
   - All-red window pops up, 'snapshot' creates and rgb#.jpeg file, error message in terminal 'Error transferring SPI packet'

To troubleshoot I have:
- Carefully removed and put back the Lepton module from its breakout board.
- Placed the CS pin in #24 (CE0) and #26 (CE1) and matched it correspondingly for spidev in the LeptonThread.cpp script (0.0 for CE0, 0.1 for CE1)
- From an answer in this thread, the wiring guide I've followed is allegedly false. So I experimented my full pin layout between these two patterns:
     - [SPI0] MOSI: #19, MISO: #21, SCLK: #23, CE0: #24, CE1: #26
     - [SPI1] MOSI: #38, MISO: #35, SCLK: #40, CE0: #12, CE1: #11
- Set DEBUG_LEPTON to true in the Lepton_thread.h script, then back to false but commented out the debug defines surrounding line 48, and changed 'sequence' to 'sequencex'
- Changed the SPI speed to 18000000 from 10000000
- Checked the voltage through the Lepton breakout board while plugged in to make sure it isn't fried (its shutter also goes off every few mins).

In addition to LeptonModule (C/C++) I have also tried the same with pylepton (Python) with identical results.

Any insights and advice are appreciated in advance

hanifizzudinrahman, 2019-12-08:
thx for the solution
after this, i will try it...

hanifizzudinrahman, 2019-12-09:
btw, i want to ask
i have done with 
- Carefully removed and put back the Lepton module from its breakout board.
(sometimes it doesn't run and sometimes it runs)
- Placed the CS pin in #24 (CE0)
- [SPI0] MOSI: #19, MISO: #21, SCLK: #23, CE0: #24,
- Changed the SPI speed to 18000000 from 10000000
(in SPI.cpp) `unsigned int spi_speed = 18000000;` and `spi_speed = 18000000;`

in yours, we must
**Set DEBUG_LEPTON to true in the Lepton_thread.h script, then back to false but commented out the debug defines surrounding line 48, and changed 'sequence' to 'sequencex'**

but, in mine there is no Lepton_thread.h but LeptonThread.h and the code is
````
#include "LeptonThread.h"

#include "Palettes.h" 
#include "SPI.h"
#include "Lepton_I2C.h"

#define PACKET_SIZE 164
#define PACKET_SIZE_UINT16 (PACKET_SIZE/2)
#define PACKETS_PER_FRAME 60
#define FRAME_SIZE_UINT16 (PACKET_SIZE_UINT16*PACKETS_PER_FRAME)
#define FPS 27;
 
LeptonThread::LeptonThread() : QThread()
{
}

LeptonThread::~LeptonThread() {
}

void LeptonThread::run()
{
	//create the initial image
	myImage = QImage(80, 60, QImage::Format_RGB888);

	//open spi port
	SpiOpenPort(0);

	while(true) {

		//read data packets from lepton over SPI
		int resets = 0;
		for(int j=0;j<PACKETS_PER_FRAME;j++) {
			//if it's a drop packet, reset j to 0, set to -1 so he'll be at 0 again loop
			read(spi_cs0_fd, result+sizeof(uint8_t)*PACKET_SIZE*j, sizeof(uint8_t)*PACKET_SIZE);
			int packetNumber = result[j*PACKET_SIZE+1];
			if(packetNumber != j) {
				j = -1;
				resets += 1;
				usleep(1000);
				//Note: we've selected 750 resets as an arbitrary limit, since there should never be 750 "null" packets between two valid transmissions at the current poll rate
				//By polling faster, developers may easily exceed this count, and the down period between frames may then be flagged as a loss of sync
				if(resets == 750) {
					SpiClosePort(0);
					usleep(750000);
					SpiOpenPort(0);
				}
			}
		}
		if(resets >= 30) {
			qDebug() << "done reading, resets: " << resets;
		}

		frameBuffer = (uint16_t *)result;
		int row, column;
		uint16_t value;
		uint16_t minValue = 65535;
		uint16_t maxValue = 0;

		
		for(int i=0;i<FRAME_SIZE_UINT16;i++) {
			//skip the first 2 uint16_t's of every packet, they're 4 header bytes
			if(i % PACKET_SIZE_UINT16 < 2) {
				continue;
			}
			
			//flip the MSB and LSB at the last second
			int temp = result[i*2];
			result[i*2] = result[i*2+1];
			result[i*2+1] = temp;
			
			value = frameBuffer[i];
			if(value > maxValue) {
				maxValue = value;
			}
			if(value < minValue) {
				minValue = value;
			}
			column = i % PACKET_SIZE_UINT16 - 2;
			row = i / PACKET_SIZE_UINT16 ;
		}

		float diff = maxValue - minValue;
		float scale = 255/diff;
		QRgb color;
		for(int i=0;i<FRAME_SIZE_UINT16;i++) {
			if(i % PACKET_SIZE_UINT16 < 2) {
				continue;
			}
			value = (frameBuffer[i] - minValue) * scale;
			const int *colormap = colormap_ironblack;
			color = qRgb(colormap[3*value], colormap[3*value+1], colormap[3*value+2]);
			column = (i % PACKET_SIZE_UINT16 ) - 2;
			row = i / PACKET_SIZE_UINT16;
			myImage.setPixel(column, row, color);
		}

		//lets emit the signal for update
		emit updateImage(myImage);

	}
	
	//finally, close SPI port just bcuz
	SpiClosePort(0);
}

void LeptonThread::performFFC() {
	//perform FFC
	lepton_perform_ffc();
}
```

```
and also, there are no **DEBUG_LEPTON**  and 
**'sequence'**
 or 
**'sequencex'**
```

Henrikcox, 2019-12-09:
The LeptonThread.h file I'm referring to is in the raspberrypi_qt folder (under software). You have the wrong file open. 

Follow the third comment's example in [this thread](https://groups.google.com/forum/#!topic/flir-lepton/GWHD1KMVYaE). Hopefully that works for you; I still have not been able to get mine working.

hanifizzudinrahman, 2019-12-10:
ohhhhh, im in wrong file. ok thx, i'll try it
btw, what the different between when processing raspberrypi_video and raspberrypi_qt?

hanifizzudinrahman, 2019-12-10:
i have try it but still wrong when try raspberrypi_qt
the message error is "Error transfering SPI Packet"
i have modified the code in **LeptonThread.cpp**
- const char *LeptonThread::device = "/dev/spidev0.0";
so, im using [SPI0] MOSI: #19, MISO: #21, SCLK: #23, CE0: #24,
- unsigned int LeptonThread::speed = 18000000;
before the speed is 16000000
in **LeptonThread.h**
- std::list< std::pair<int, int> > sequencex;
before is sequence

[image]

Henrikcox, 2019-12-10:
Yes that is one of the errors I'm getting right now too. Still have not found a solution to it I'm afraid.

hanifizzudinrahman, 2019-12-11:
i use a new code from Frank

[LEPTON 2.zip](https://github.com/groupgets/LeptonModule/files/3948035/LEPTON.2.zip)

I will say full thank you to you because, with this code, is absolutely make me happiest person in the world, hehehe...

But, there is little problem in my project (**Lepton 2.5** with board V1.4 + **Raspberry Pi 3 B**)
my schematic is

- CS = SPI0 CS0 (Pin 24)            # **Use spidev0.0**
- MOSI = SPI0 MOSI (Pin 19)
- MISO = SPI0 MISO (Pin 21)
- CLK = SPI0 SCLK (Pin 23)
- GND = GND (Pin 14)
- VIN = 3.3V PWR (Pin 17)
- SDA = I2C1 SDA (Pin 3)
- SCL = I2C1 SCL (Pin 5)

According with the suggestion above, i make some change in ....
* **LeptonThread.cpp**
- const char *LeptonThread::device = "/dev/spidev0.0";   # (can change in spidev0.1)
- unsigned int LeptonThread::speed = 18000000;            # (from 16000000)
- qDebug() << resets << "reset," << errors << "errors";    # (before this line is commented)
* **LeptonThread.h**
- #define DEBUG_LEPTON true                                       # (before this line is false)

when i run the Lepton-CaptureRaw, the result is good like this, but... 
[image]

but..., sometimes when i try **10 times run this** i can video it the thermal video and when i press the button "Snapsot" it will give me the .jpg picture (1-rgb.jpg, 2-rgb.jpg) and .bin format (1-raw.bin, 2-raw.bin) that this raw format will show a temperature value (i'm not yet with this) but **2 times will fail**. 
The error is "**Error transfering SPI packet**" and the **layer full of red**. Can you help me with this? why this happen?

Henrikcox, 2019-12-11:
I'm glad you're making progress with it. Is the LEPTON 2.zip file the same as the our_video.zip file from the top of that thread (https://groups.google.com/forum/#!topic/flir-lepton/GWHD1KMVYaE)? 

I was following the same thing as you did but still get the red screen and either "Error transferring SPI packet" or "Packet reset counter hit 750." But I am running raspberrypi_video with 
`sudo ./raspberrypi_video`

Are you running a different file, how are you running it? I am not seeing where Lepton-CaptureRaw that you're running is.

gholibeigian, 2021-07-01:
Hi,
It seems very crazy, but when I changed the jumper wires with the short ones(shortest), I did not get this error anymore!

norton-chris, 2024-10-22:
Has this been figured out? I'm experiencing the same issues on each executable in the original post. I'm testing on the Lepton 3.1R.