groupgets/LeptonModule #73: v4l2lepton for Lepton 3.5

istranic, 2020-02-19:
Just wondering if anyone has updated v4l2lepton to work with Lepton 3.5?

kekiefer, 2020-02-19:
Not to my knowledge, but it would be nice to have. There is a PR #71 (that I still have to review) that builds upon some previous work that breaks Lepton 3 support into a common module.

istranic, 2020-02-19:
@kekiefer Thanks for the info. It doesn't seem like PR #71 is related to v4l2lepton, but are you saying that some of the work in #71 could be re-used in order to make v4l2lepton work with Lepton 3.5?

kekiefer, 2020-02-19:
The second thing. #71 has nothing to do with v4l2lepton directly, but it creates common code that could be reused by v4l2lepton to support Lepton 3.

istranic, 2020-02-20:
Cool, thx

crisposada, 2020-12-04:
Hi @istranic ! Have you had progress on this? I would like to get v4l2lepton to work with Lepton 3.5

istranic, 2020-12-04:
@crisposada No sir, still hoping someone gets this to work.

ledfreak3d, 2023-02-11:
2 years later and nobody ?

marty-sullivan, 2023-02-11:
@ledfreak3d I wrote v4l2lepton. The difference between frame grabbing in Lepton v2 and v3 is significant. However, the code is there in:

https://github.com/groupgets/LeptonModule/blob/master/software/raspberrypi_video/LeptonThread.cpp

If someone can basically take this code and modify v4l2lepton code to match here:

https://github.com/groupgets/LeptonModule/blob/master/software/v4l2lepton/v4l2lepton.cpp

This could provide the solution. Unfortunately, I do not have a Lepton 3+ and so I cannot update the code myself.

ledfreak3d, 2023-02-11:
Ill see what i can do

prismalink12, 2023-03-16:
@ledfreak3d Hey did you ever upgrade the code for 3.5?