groupgets/LeptonModule #85: V4l2lepton stream to RPi IP

Cherry97Pepis, 2021-07-16:
Hi, 
I'm trying to stream the thermal video (Lepton 2.5) with Raspberry Pi 4B and get access to it in my local network. I install v4l2looback, modprobe done, /dev/video0 made, but no image i can see in VLC when I open as capture device.  Additionaly, when i run v4l2lepton it prints "waiting for a sink" and nothing more. 
Have you any ideas how to fix it? 
I was trying to change the CS pin, stream through GStreamer, with Motion, but I can't do that right, i think. 
I'm working on it for a two month and now I still don't know what is wrong. 
Please, help me if u know how to fix it

Cherry97Pepis, 2021-07-16:
Problem fixed, there was a bug in SPI connection :) I don't know how to repair it so i wrote the code again

closed 2021-07-16 by Cherry97Pepis

Cullpepper, 2021-07-16:
Glad to hear it, good luck with your project.

Cherry97Pepis, 2021-07-16:
Thanks, now time to start working on Motion to stream video :) 
Thanks for help