groupgets/LeptonModule #87: What is FFC, and why does it do nothing?

Tylersuard, 2021-09-17:


BehicKlncky, 2022-08-26:
FFC is the calibration that setting the camera for more accurate IR vision. It needs to done by 15-20 seconds periods.

marty-sullivan, 2022-08-26:
@Tylersuard to expand on the other comment, FFC especially needs to be used when the IR sensor is in a fixed position and pointed at the same scene for extended periods of time. You typically don't need to run FFC routines yourself as the Lepton automatically does the recalibration routine by default every 15 or so seconds. 

FFC only applies to sensors with a shutter (most have the shutter). You'll notice the FFC routine since the Lepton will make a "click" noise whenever the shutter closes to calibrate.

What is physically happening during FFC is, the shutter closes in front of the IR sensor and allows the camera to compare its own temperature relative to the temperature of the scene. Since the temperature of the sensor itself relative to the scene is an important aspect of its reading, this allows for better readings.